HeyWillow/willow #259: Failed startup: failed to open NVS namespace WIFI: ESP_OK

myxor, 2023-09-07:
After a fresh `flash-erase` and then `flash` the startup fails:

```
[09:15:41.816] tio v2.6
[09:15:41.816] Press ctrl-t q to quit
[09:15:41.817] Connected
I (181) esp_image: segment 1: paddr=00083210 vaddr=3fc9f6c0 size=06c64h ( 27748) load
I (187) esp_image: segment 2: paddr=00089e7c vaddr=40378000 size=0619ch ( 24988) load
I (193) esp_image: segment 3: paddr=00090020 vaddr=42000020 size=148780h (1345408) map
I (437) esp_image: segment 4: paddr=001d87a8 vaddr=4037e19c size=11518h ( 70936) load
I (461) boot: Loaded app from partition at offset 0x30000
I (492) boot: Set actual ota_seq=1 in otadata[0]
I (492) boot: Disabling RNG early entropy source...
I (503) opi psram: vendor id : 0x0d (AP)
I (503) opi psram: dev id    : 0x02 (generation 3)
I (503) opi psram: density   : 0x03 (64 Mbit)
I (507) opi psram: good-die  : 0x01 (Pass)
I (512) opi psram: Latency   : 0x01 (Fixed)
I (517) opi psram: VCC       : 0x01 (3V)
I (521) opi psram: SRF       : 0x01 (Fast Refresh)
I (527) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (532) opi psram: BurstLen  : 0x01 (32 Byte)
I (537) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (543) opi psram: DriveStrength: 0x00 (1/1)
I (549) spiram: Found 64MBit SPI RAM device
I (553) spiram: SPI RAM mode: sram 80m
I (557) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (564) cpu_start: Pro cpu up.
I (568) cpu_start: Starting app cpu, entry point is 0x403796cc
I (0) cpu_start: App cpu up.
I (860) spiram: SPI SRAM memory test OK
I (869) cpu_start: Pro cpu start user code
I (869) cpu_start: cpu freq: 240000000
I (869) cpu_start: Application information:
I (871) cpu_start: Project name:     willow
I (876) cpu_start: App version:      1
I (881) cpu_start: Compile time:     Sep  7 2023 06:53:32
I (887) cpu_start: ELF file SHA256:  c900ff9308aeb52c...
I (893) cpu_start: ESP-IDF:          v4.4.5-dirty
I (898) cpu_start: Min chip rev:     v0.0
I (903) cpu_start: Max chip rev:     v0.99 
I (908) cpu_start: Chip rev:         v0.1
I (913) heap_init: Initializing. RAM available for dynamic allocation:
I (920) heap_init: At 3FCAC988 len 0003CD88 (243 KiB): D/IRAM
I (926) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (933) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (939) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (948) spi_flash: detected chip: gd
I (951) spi_flash: flash io: dio
I (956) sleep: Configure to isolate all GPIO pins in sleep state
I (962) sleep: Enable automatic switching of GPIO sleep configuration
I (969) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (985) spiram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (00:16:20.585) WILLOW/MAIN: Starting up! Please wait...
I (00:16:20.859) WILLOW/MAIN: Waiting on SPIFFS mount...
I (00:16:21.359) WILLOW/MAIN: SPIFFS mounted
I (00:16:21.359) WILLOW/CONFIG: opening /spiffs/user/config/willow.json
E (00:16:21.482) WILLOW/CONFIG: failed to open /spiffs/user/config/willow.json
E (00:16:24.152) WILLOW/MAIN: failed to open NVS namespace WIFI: ESP_OK

```

nikito, 2023-09-07:
Was this flashing via WAS? Or did you just go directly to the flashing website? If in WAS, did you run through all the configs first before flashing?

stintel, 2023-09-08:
Your NVS partition does not have an NVS namespace, and you reference commands like `erase-flash` and `flash`, so you are most likely not using Willow Web Flash. You should follow [the quick start guide](https://heywillow.io/quick-start-guide/).

Alternatively, if you do not want to use Willow Web Flash, you can write the NVS partition manually. Instructions are in commit 36ee1f70bf1d23d2cf654afbd2a95db448e73699 and 021bb7ad08244732167c1099d44399da1b80bda5.

myxor, 2023-09-11:
I used the guide described at https://heywillow.io/development/ because this was the way i flashed it the first time (before WAS existed).
Using WAS the flash process worked. Thanks for the help!

closed 2023-09-12 by myxor