HeyWillow/willow #262: Need to resart ESP Box after few commands are recognized successfully

pk802, 2023-09-12:
I am getting this error after few successful commands. 

[image]
[image]

kristiankielhofner, 2023-09-12:
The hosted server you are connecting to (infer.tovera.io) has served over 1,000 requests today with no logged connection errors according to metrics from the Cloudflare edge.

You are likely experiencing some connectivity issue(s) with your local network, internet connection, or somewhere in between.

This is going to be very difficult to diagnose because it involves some really fundamental things and a ton of variables.

kristiankielhofner, 2023-09-12:
Are you using AMR-WB from the advanced settings?

pk802, 2023-09-12:
Yes, I was using AMR-WB from the advanced setting. I changed it back to PCM. 
Having same issue. Works for 2 commands and then same error.

pk802, 2023-09-12:
I tried https://wisng.tovera.io/api/willow?model=large&beam_size=5
as it was mentioned in few other issue and it seem to be working fine.

https://github.com/toverainc/willow/issues/200#issuecomment-1605240757

nikito, 2023-09-12:
Have you tried running a continuous ping to the inference url to see if any drops are happening that could cause the socket to reset? Could leave it up and see if you are getting drops when the command fails. You can also run a continuous ping to the willow device to see if it is being dropped from WiFi for some reason; some APs will drop devices off they after before a minimum rssi.

Hope this helps! 🙂

stintel, 2023-09-13:
Also which version are you running?

kristiankielhofner, 2023-09-13:
> I tried https://wisng.tovera.io/api/willow?model=large&beam_size=5 as it was mentioned in few other issue and it seem to be working fine.
> 

This host is the exact some server (at a different hostname) with a different configuration for the speech recognition. The error you are getting comes from the Willow Inference Server when it fails to load the audio from your Willow device - before speech recognition configuration like model, beam_size, etc. None of these things should have any effect. I asked about AMR-WB because it's a different audio codec with additional steps in Willow and WIS to load the audio. It's under "Advanced Settings" because it should only be used in very special circumstances. We are working on updating the web interface to better explain the various settings.

As @nikito said your issues are almost certainly network related. We use a tried and true Wifi implementation from Espressif and HTTP/HTTPS (TCP) to transport the audio so it should arrive reliably to WIS even under challenging conditions. There is likely something specific to your environment that's especially challenging and causing the audio transfer to get corrupted, truncated, etc.

pk802, 2023-09-24:
> Also which version are you running?

@stintel I am using latest code.

> Have you tried running a continuous ping to the inference url to see if any drops are happening that could cause the socket to reset? Could leave it up and see if you are getting drops when the command fails. You can also run a continuous ping to the willow device to see if it is being dropped from WiFi for some reason; some APs will drop devices if they are below a minimum rssi.
> 
> Hope this helps! 🙂

@nikito I ran continuous ping to inference URL and to my ESP32 box. There was no drop in packets. 

This problem still persist. Also I observed that after device is idle for few hours it dosen't work. I am sharing screenshot where you can see timestamp. I starts working again if I restart the ESP32 Box. 
[image]

nikito, 2023-09-24:
Are you running the latest release of willow and WAS? I feel like this was reported a while back and solved. If so, it may be that something is wrong with this box, maybe it needs a full reflash? I've been running 3 boxes 24/7 and have not had any of them lock up or display an error like that running the latest rc1 build. 😊

pk802, 2023-09-24:
> Are you running the latest release of willow and WAS? I feel like this was reported a while back and solved. If so, it may be that something is wrong with this box, maybe it needs a full reflash? [...]

All my boxes were on rc1.  Now I have updated my 3 boxes to rc2 and WAS. Let me see if I get same errors.

kristiankielhofner, 2023-10-08:
@pk802 - Have you tried again with our latest release?

stintel, 2025-02-23:
We believe this is solved, and no feedback for > 1 year. Closing.

closed 2025-02-23 by stintel