HeyWillow/willow #268: ESP32-S3-BOX-3 not booting

ashishpandey, 2023-09-23:
I followed the BOX-3 support added via this MR: https://github.com/toverainc/willow/pull/263

tried flashing using https://feature-esp32-s3-box-3.willow-web-flash.pages.dev/

flashing seems to go well, until the resetting device message, but nothing happens after that
```
Wrote 16777216 bytes (4914697 compressed) at 0x0 in 88.335 seconds.
Hash of data verified.
Leaving...
Flash successful! Resetting your device.
```

no display, no boot. If I disconnect and try to connect again, the flasher connects but then just freezes. The only way to recover is to use ESP Launcher to flash stock demo firmware again. I have tried multiple time, with same result

Any ideas how I can troubleshoot further?

SamJongenelen, 2023-09-23:
I've been having similar issues. It happened to me when I used the wrong WAS url. The device cant boot (and wont show it on display). Because the logs on the webflasher dont work, you cannot guess what is wrong :D 

try this flasher: https://feature-serial-output.willow-web-flash.pages.dev/   - it worked better for me to get logs. 

So regarding your issue;
Do you have WAS running on local subnet? I do, the URL format that works for me is `ws://192.168.1.41:8502/ws`

ashishpandey, 2023-09-24:
> Do you have WAS running on local subnet?
thanks @SamJongenelen, it was probably this. the device Wifi was on a different VLAN to the WAS. I had to add a firewall rule to get past that

I thought these kind of issue would be post boot, so never thought to bother about network configuration at flash stage

Also, thanks for the flasher, very useful. perhaps should be an option on webflasher too :)

All good, I've made progress now. closing this

closed 2023-09-24 by ashishpandey