HeyWillow/willow #270: Dutch language support?

supremelucifer, 2023-09-23:
Hi I just got the box and using willow in Home Assistant. When i turn the language to dutch, it doesn'r recognize the entities and shows my voice input in english. When i set the language to english in Hassio it is working.

Does it not work with the dutch language?

supremelucifer, 2023-09-23:
Sorry, i already got it

nikito, 2023-09-23:
Are you setting the language willow uses for stt to dutch as well?

SamJongenelen, 2023-10-01:
> Are you setting the language willow uses for stt to dutch as well?

Can we do this for the cloud version too? I've not seen settings for this

nikito, 2023-10-01:
I believe you can modify the stt url and add ?language= and put your 2 letter language code

kristiankielhofner, 2023-10-02:
This has been changed in newer WAS releases (including our hosted endpoint) to use `?force_language=` with your 2 letter language code in your WIS URL.

We have other users using dutch and once configured it seems to work well.

We will be working to address this with our new and completely revamped WAS Web UI. It isn't implemented yet but an upcoming release will have drop down/checkbox/etc elements to assist you in configuring WIS options without having to know all of the URL parameter options.

closed 2023-10-10 by kristiankielhofner