HeyWillow/willow #271: Duckdns/LE Cert Issue

longunmin, 2023-09-24:
I have everything set up properly, but I use a duckdns address along with LE certificate. When I try and run a command, I'm giving an HTTP 404 error. My assumption is that duckdns/le isn't allowed with willow because I've heard it requires extra processing/memory for those certs.  Would it be possible to make that combination usable, as I know that is a popular setup for HA users

nikito, 2023-09-25:
404 error usually means the server couldn't be resolved, is there a possibility that the hostname isn't reachable on the network willow is on? Also do you have tls turned on in config?

longunmin, 2023-09-25:
I am able to access the hostname on my network.  I do have tls on.  In the `Home Assistant Host` field I have ` xxxxx.duckdns.org` and then `Home Assistant Port` of `443`.  If I got to `https://xxxxx.duckdns.org:443` I'm directed to my home assistant frontend.

nikito, 2023-09-25:
Do you get a reply if going to /api/ on that url? Also have you tried just using the local address of your HA instance? Since the traffic never leaves the local network (willow talks to HA through a direct websocket) there's no need to use the public hostname. 😊

nikito, 2023-09-25:
Probably also should have asked this first, what version of Home Assistant are you running?

kristiankielhofner, 2023-09-25:
@longunmin - Thanks for providing your Home Assistant config settings - do you have TLS enabled in addition to port 443 and host? Seeing as you're getting 404 you almost certainly are but it can't hurt to confirm.

HTTP 404 means the host could be resolved and connects successfully but the path was not found on the server, which is strange as you've already validated it works from your browser. Are your Willow devices on the same network as the machine you can connect to to HA with your browser?

longunmin, 2023-09-25:
@nikito If I just use `https://xxxx.duckdns.org:443/api/` I get 401 unauthorized.  However, if I use a rest command with the url and authorization using an HA long lived access token, I get a valid status code.  For HA, I am forwarding my ports (8123/443) and using duckdns as the external url, so that I can have remote access.  This means that the internal IP doesn't resolve to HA anymore, so I can't use the Internet IP for willow.  This could be achieved using a reverse proxy (NGINX), but if possible, I'm trying to avoid that, as I worry about breaking other parts of my setup.  I am on HA 2023.1.7 running in docker.

@kristiankielhofner yes both the willow and HA are on the same network.  And yes, I have TLS enabled

nikito, 2023-09-25:
Yes I think at minimum 2023.5 is needed, though it may not hurt to go to the newest version. Looking at the HA blog I think 2023.5 was the earliest version they added this API (Think this was the first release in the Year Of Voice). :)

longunmin, 2023-09-25:
Perfect. I will aim to get to 2023.5 and report back.

kristiankielhofner, 2023-09-25:
Note it's 2023.5 or later. If you're going through the effort to upgrade you should try to get to the most recent release supported by your configuration.

longunmin, 2023-09-28:
@nikito @kristiankielhofner I upgraded to 2023.5.6 but I am still getting the 404 error

nikito, 2023-09-28:
Have you set up an assistant in HA? If you go to your lovelace interface, do you see the assist icon?
[image]
[image]
[image]

kristiankielhofner, 2023-09-28:
Is there a proxy of some sort in between Willow and HA? Something like nginx, etc?

Try running this:

`curl -v -X POST https://YOUR_HOSTNAME_AND_PORT/api/conversation/process`

longunmin, 2023-09-28:
@kristiankielhofner ok that fixed it. Adding in the assist config.  Now it is just giving me no matching HA intent, but that's because I haven't build anything up yet.  So was never a duckdns issue.  Thank you!

kristiankielhofner, 2023-09-28:
Sometimes it's the obvious, I guess. Thanks @nikito !

closed 2023-09-28 by kristiankielhofner

nikito, 2023-09-28:
> @kristiankielhofner ok that fixed it. Adding in the assist config. Now it is just giving me no matching HA intent, but that's because I haven't build anything up yet. So was never a duckdns issue. Thank you!

Just for your information, there's built in intents in HA that should work "out of the box", such as turn on/off lights and areas. However for them to work, you have to go into the Expose tab of your assistant settings and expose those devices to the assistant. I think by default nothing is exposed unless you choose to make it exposed. :)