HeyWillow/willow #277: Updated Willow (though WAS) and also WIS on server.

bert269, 2023-09-26:
Then I updated to the latest (willow-RC2) software on each of the ESP32-S3 (and one S3-lite) device and none is responding. They respond to the wakework, but just sits there. This is a screenshot from what's happening on WIS:
[image]

I'll flash the previous RC-1 to the first S3 and see what happens. All the willow were working fine still this morning, util I updated everything.

bert269, 2023-09-26:
I re-installed RC-1 (OTA) to the S3-lite device.
No change,. So I rebooted the WIS server and this was the result after I issued the wake-word with a command:
[image]

Then a few moments later. I saw this on the WIS server display:
[image]

bert269, 2023-09-26:
Stopped WIS and WAS.
Then I run sudo apt update && sudo apt upgrade -y and it had 48 packages to install (I just did this two days ago). Included were some Nvidia pacakges - so maybe, just maybe, I had older code for a newer WIS?
Rebooted the server.
Started WAS and saw the clients.
Re-installed WIS - as I am not sure with the older NVidia code (from before OS upgrade), what happend earlier when I tried to install it. It seems to me that more code is being downloaded during the WIS re-install.

Nothing - no response from the S3 devices.

bert269, 2023-09-26:
The PROBLEm is the BEAM size....! a while back someone on the github response wold me to use BEAMsize=5. Well, starting up the server, I noticed it uses BEAM-SIZE=1. So I modified my config, flash the S3-lite with beam-size=1 and it magically worked.

WHAT is the difference - why did the beam size owrked before with 5 and not it requires 1?
AT elast it is working and I am happy again.

kristiankielhofner, 2023-09-26:
We've never made an API change between Willow and WIS so I'm not sure why anything with the releases should matter.

Beam size tweaks a parameter inside the model more-or-less regarding accuracy. People throw a lot of things around on Github, the internet, etc but the medium model with beam_size=1 is the current WIS default. We've been doing extensive testing and the next release of WIS will actually default to small beam_size=2. That said we also routinely test all of the possible model and beam_size configurations.

I've never seen and we've never had reports of there being issues with any of the models or beam sizes, the only thing I can think of is larger models and larger beam sizes use more GPU memory. What model of Nvidia GPU do you have? We do a lot on WIS startup to define parameters for even really low VRAM and older GPUs, I myself extensively test with a GTX 1060 6GB which is towards the lowest GPU we could support.

Also, you probably want to be a little more careful randomly upgrading Nvidia drivers. We have extensive compatibility with Nvidia drivers as well (we test with 525-535 on Ubuntu) but like all driver upgrades there can definitely be lower-level difficult to debug issues from swapping random drivers around.