HeyWillow/willow #287: Esp32-S3 Korvo-1

EelcoBaartse, 2023-10-01:
Although officially unsupported, would the Esp32-S3 Korvo-1 board work with Willow? I prefer not to have a (touch)screen, so this would be great. (And the device is actually in stock...)

kristiankielhofner, 2023-10-08:
Technically speaking anything based on the ESP32-S3 could work. However, for this board specifically it isn't supported by the [Espressif Audio Development framework](https://github.com/espressif/esp-adf) Willow uses.

So a first step would be support by the ESP ADF Audio HAL and then some changes would be required in WIllow to conditionally handle things like conditionals for the display, etc.

We also have concerns about development boards generally. One of the things that contributes to high audio quality is acoustic tuning of the enclosure. The dev board itself likely wouldn't capture audio that well (especially far field) and without a properly designed enclosure it's likely only to get worse. For these reasons we're reluctant to support audio development boards directly.

Regarding the ESP-BOX-3, the initial stock was a limited pre-production run by Espressif. We've been told that the final production run with injection molded enclosure should be available within a couple of weeks.

janstadt, 2024-04-15:
Just curious if anything has progressed here with supporting the korvo (i have a couple v1.1s) boards? I am currently testing microWakeWord on a korvo and its working fairly well, but i really like some of the features willow has like auto-correct. Would be great if willow had support for non S3 based architectures.  https://github.com/esphome/feature-requests/issues/2430#issuecomment-2039242357

potelux, 2024-04-23:
> Just curious if anything has progressed here with supporting the korvo (i have a couple v1.1s) boards? I am currently testing microWakeWord on a korvo and its working fairly well, but i really like some of the features willow has like auto-correct. Would be great if willow had support for non S3 based architectures. [esphome/feature-requests#2430 (comment)](https://github.com/esphome/feature-requests/issues/2430#issuecomment-2039242357)

There are no current plans to support non-S3 based architectures. This is what the [documentation](https://heywillow.io/hardware/#other-esp32-s3-based-boards) says:
> The ESP32 S3 is what really "makes the magic happen" for Willow as the high speed PSRAM enables a lot of the more advanced neural features. It is certainly possible that Willow can run on other ESP32-S3 based boards but supporting these is up to you!

> In the future we may officially support other ESP32-S3 (or successor series) based boards but they are not supported at this time.

janstadt, 2024-04-23:
The korvo v1.1 (and others) have PSRAM. I have microwakeword running on v1.1 and was hoping to alternatives like willow to compare.