HeyWillow/willow #288: Willow audio response type 'Chimes' does not work

SamJongenelen, 2023-10-02:
Hello, 

I am using a ESP32-S3-BOX to run Willow.  Using `Willow audio response type`  `Chimes` and it gives me no sounds.  Commands are executing fine. TTS works fine.

I am not sure this is a known issue and I've not been able to get chimes to work, but now `0.1.0` is live I am thinking it should work. 

I can provide logs - but the device is currently in use in the living room  - and i think i need to reflash to see logs, right?

kristiankielhofner, 2023-10-02:
This is not a known issue and we haven't had any reports of it. I myself use and test chime responses regularly.

You don't need to reflash to see logs, you can connect the device to the web flasher and start the serial terminal or use a serial terminal on any machine. If you could do this and provide the log output here that would be very useful.

There are many ways to run a serial console, the prefered and officially supported way is to use `tio`on a host machine and the `./utils.sh monitor` function from our source tree as documented [here](https://heywillow.io/development/). Note that you do not need to install the full build environment or configure and build Willow yourself. Tio on your machine and a clone of the willow repo is all you need for serial monitor.

SamJongenelen, 2023-10-02:
Great, thanks will do an circle back. All console options are USB right? Would be nice in WAS too

kristiankielhofner, 2023-10-02:
Your operating system should detect the USB serial device and make it a serial port. On Linux this is a `/dev/ttyACM*` device, on Mac it is `/dev/cu.usbmodem*` and on Windows I think it's referred to as a "USB serial composite device" or something like that (but it's basically just a COM port).

We have investigated streaming logs to WAS, there are some potential issues with this but once we get around to digging deeper we will have to look at what the network traffic, load on the device, and UI/UX looks like. If anything it would be an advanced configuration option when/if available.

SamJongenelen, 2023-10-04:
Ok - hooked up the box and immidiately the box chimed. So no info found in the logs, weird.