HeyWillow/willow #289: Willow not using HA response for TTS

ashishpandey, 2023-10-02:
I am running the following:
* **_willow_** version `will-be-deleted-box-3` on `ESP32-S3-BOX-3`
* **_WAS_** image `main` 
* **_WIS_** running locally as per build instructions
* **_Home Assistant_** version `2023.9.3`, with some custom sentences and intent_scripts to test an interaction

They seems to working well, in the sense
* willow understands my sentence
* willow calls WIS and gets correct STT response back
* willow calls HA, and HA responds with correct intent_script

What doesn't seem to be working well is the TTS part. While the HA response has expected speech text, willow seems to be sending a different text (`ok`) to WIS for TTS

relevant logs below:
```
I (09:25:46.624) WILLOW/MAIN: Startup complete! Version: will-be-deleted-box-3. Waiting for wake word.
I (09:25:52.691) WILLOW/AUDIO: AUDIO_REC_WAKEUP_START
I (09:25:53.477) WILLOW/AUDIO: AUDIO_REC_VAD_START
I (09:25:53.478) WILLOW/AUDIO: WIS HTTP client starting stream, waiting for end of speech
I (09:25:53.479) WILLOW/AUDIO: Using WIS URL 'https://wis.******.com/api/willow'
I (09:25:55.433) WILLOW/AUDIO: AUDIO_REC_VAD_END
I (09:25:55.434) WILLOW/AUDIO: AUDIO_REC_WAKEUP_END
I (09:25:55.458) WILLOW/AUDIO: WIS HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (09:26:03.572) WILLOW/AUDIO: WIS HTTP client HTTP_STREAM_FINISH_REQUEST
I (09:26:03.573) WILLOW/AUDIO: WIS HTTP Response = {"language":"en","text":"What is living room temperature?"}
I (09:26:03.580) WILLOW/HASS: HASS URL: http://******:8123/api/conversation/process
I (09:26:03.589) WILLOW/HASS: sending '{"language":"en","text":"What is living room temperature?"}' to Home Assistant API on 'http://******:8123/api/conversation/process'
I (09:26:03.712) WILLOW/HTTP: HTTP status='200' content_length='230'
I (09:26:03.713) WILLOW/HASS: home assistant response_type: action_done
I (09:26:03.715) WILLOW/AUDIO: Using WIS TTS URL 'https://wis.******.com/api/tts?format=WAV&speaker=CLB&text=ok'
I (09:26:04.146) WILLOW/HASS: HTTP POST response body:
{
        "response":     {
                "speech":       {
                        "plain":        {
                                "speech":       "Living Room temperature is 25.4 degrees",
                                "extra_data":   null
                        }
                },
                "card": {
                },
                "language":     "en",
                "response_type":        "action_done",
                "data": {
                        "targets":      [],
                        "success":      [],
                        "failed":       []
                }
        },
        "conversation_id":      null
}
```

the intent script in HA looks like this:
```yaml
LivingRoomTemperature:
  speech:
    text: "Living Room temperature is {{ states('sensor.living_room_sensor_temperature') }} degrees"
```

**_Is there a way willow can use speech returned from HA?_** 
Is this misconfiguration on my part, incompatibility between willow & HA, or a known issue at this point?

nikito, 2023-10-02:
Have you enabled TTS replies in your willow configs?
in WAS, if you go to your general settings, there's a dropdown for Willow Audio Response Type, you would need to ensure this is set to "Text to Speech" and also provide the URL to your WIS instance's TTS path (by default the tovera instance's TTS url is supplied).
[image]
Hope this helps!

ashishpandey, 2023-10-02:
Yes, TTS is enabled in willow config. See this part in the logs I posted

```
I (09:26:03.715) WILLOW/AUDIO: Using WIS TTS URL 'https://wis.******.com/api/tts?format=WAV&speaker=CLB&text=ok'
```

I have looked a bit further and noticed that HAAS integration in willow uses 2 different approaches depending on whether assist_pipeline is enabled in Home Assistant
* without `assist_pipeline`, HTTP POST is used and TTS does not used the returned speech as in my first post
* with `assist_pipeline` enabled, websockets are used, and in this case, correct TTS response is used. See below logs with assist_pipeline enabled
```
I (08:00:51.619) WILLOW/AUDIO: WIS HTTP Response = {"language":"en","text":"What is living room temperature?"}
I (08:00:51.627) WILLOW/HASS: sending command to Home Assistant via WebSocket: {
	"end_stage":	"intent",
	"id":	51,
	"input":	{
		"text":	"What is living room temperature?"
	},
	"start_stage":	"intent",
	"type":	"assist_pipeline/run"
}
I (08:00:51.763) WILLOW/HASS: home assistant response_type: action_done
I (08:00:51.766) WILLOW/HASS: received run-end event on WebSocket: {
	"id":	51,
	"type":	"event",
	"event":	{
		"type":	"run-end",
		"data":	null,
		"timestamp":	"2023-10-02T12:45:55.940650+00:00"
	}
}
I (08:00:51.779) WILLOW/AUDIO: Using WIS TTS URL 'https://wis.******.com/api/tts?format=WAV&speaker=CLB&text=Living Room temperature is 25.2 degrees'
```

I looked at the HASS integration, and I don't seem to see picking up the speech response in HTTP POST flow. Maybe that's the issue?

ashishpandey, 2023-10-02:
@kristiankielhofner Yes, it works fine with assist_pipeline enabled

But do note that HA will not auto enable pipelines. I have been using HA for a really long time, prior to when they probably introduced assist pipelines. If it's backward compatible, and installation is code managed (like mine), HA does not indicate in any way that this can be a problem, and will keep working with HTTP conversation API

I am find with assist pipeline mode, but maybe willow should either support the legacy API or clearly opt out. Right now, it tries to detect which way to go, and only works in one of those paths

kristiankielhofner, 2023-10-02:
You are correct and this is reflected in the HA documentation I provided.

One of the things we are trying to solidify (with feedback from the community) is officially supported HA versions. I think we're generally leaning towards a six month timeframe as HA moves very quickly but we're trying to get a feel for when Willow users generally upgrade their HA installs. In any case it's likely support for the conversation API will be removed completely in an upcoming release as HA itself is moving further and further away from the conversation API with their own voice/assistant support so the user experience with the conversation API is only going to get left further behind. In the meantime we will work to support TTS with the conversation API in the 0.1 series of Willow releases.

As you know there are limitless possibilities when it comes to HA installs and configurations, we can only do our best to support as many of them as possible and generally speaking adapting your install for assist pipelines is the "HA way" and more-or-less therefore the "Willow way".

ashishpandey, 2023-10-02:
I am fine with the "HA Way" or the "Willow way". Except, willow seems to try to support something that it doesn't. 

maybe simpler to warn when `hass_check_assist_pipeline` returns `has_assist_pipeline = false` and not try to `hass_post` at all. Will save people trying to figure this out some time, and be simpler on willow end as well (not having to maintain code that doesn't work)

I am good with above explanation, so closing this for now

closed 2023-10-02 by ashishpandey

longunmin, 2023-10-09:
> I am running the following:
>
> * **_willow_** version `will-be-deleted-box-3` on `ESP32-S3-BOX-3`
> * **_WAS_** image `main`
> * **_WIS_** running locally as per build instructions
> * **_Home Assistant_** [...]

Sorry for the off topic, but where/what do you set it to to use HA TTS?

kristiankielhofner, 2023-10-09:
We don't currently have support for the speech to text or text to speech interfaces exposed via HA. There are protocol support and performance issues with them.

We may look into it further in the future but Willow is intended to be used with Willow Inference Server STT and TTS.