HeyWillow/willow #296: Stuck at "Connecting to Wi-Fi..."

ggalt, 2023-10-07:
I've tried flashing my ESP32-S3-Box several times through the web interface.  All seems OK and I get "Flash successful! Reseting your device."  On the screen of the ESP32-S3-Box I get "Starting up (server)..." and "Connecting to Wi-Fi...", but it stops there.  I can see from my router, that it **has** connected to the wi-fi access point, but there is no traffic to or from the ESP32-S3-Box.  Is there a debug interface that would help me figure out what the issue is?

kristiankielhofner, 2023-10-07:
Strange!

The device has a serial console. With the device connected to your computer you can go back to the [web flasher](https://flash.heywillow.io/). Under the section marked "Serial Console" when you "Start" your browser will prompt you for the port. Select the port like you did during flashing. It will start providing serial console output. You can click "Reset" and it will restart the device so you can see logging from a fresh boot.

If the error isn't apparent you can provide the serial console output here and we'll take a look at it. Make sure not to include the configuration debug that shows your authentication settings.

ggalt, 2023-10-08:
Duh!!  I failed to open port 8502 on my firewall!  It was obvious once I had the serial console open.  I blew right past that initial page (with "start" and "reset") while flashing the device.  Sorry.

kristiankielhofner, 2023-10-08:
No problem, glad you got it sorted out!

closed 2023-10-10 by kristiankielhofner

stintel, 2023-10-10:
We have added a "Connecting to WAS ..." message on the display in a7dcbf762394c7cabd4178c019c802e1c5e6ab80 to make this less confusing.