HeyWillow/willow #301: HA Sentence Wildcards Stops Willow

longunmin, 2023-10-09:
I have tested this a few times, but it appears that when I try to call an automation in HA via a wildcard sentence, `{lists}`,  willow is unable to perform/access the intent.  I am on Willow 0.1.1 and HA 2023.8.4.  When I remove the wildcard automation, Willow goes back to working as intended

kristiankielhofner, 2023-10-09:
Willow sends the transcription as-is to Home Assistant and gets the response of whatever it does. Willow has nothing to do with specific parsing of intents, transcripts, commands, actions, etc.

Have you tried it in Home Assistant? You can click the Assist icon in the type right corner of the web interface. Type in your commands as they are transcripted. This way you can debug your intents without having to speak to Willow. When they're configured properly you can speak to Willow and we'll send the transcript that should match your intent , with the outcome being identical to typing it in the web interface.

longunmin, 2023-10-09:
Ok it appears to be a HA issue. I am trying to capture the whole sentence, but doesn't seem to want to allow me to do that

closed 2023-10-09 by longunmin