HeyWillow/willow #310: Willow fails to connect to WAS after updating

kevdliu, 2023-10-13:
Hello, 

I updated my WAS installation to the latest today with the new UI. Afterwards I flashed my willow device to 0.2.0-rc1 and it stopped being able to connect to WAS. I then tried flashing 0.1.1 (the version I had before) instead and am seeing the same issue. I verified that I am able to connect to the the WAS websocket from another computer on the LAN so I'm not sure what the issue could be. This is the console log from willow:

```
I (24) boot: ESP-IDF v4.4.5-dirty 2nd stage bootloader
I (25) boot: compile time 16:40:53
I (25) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 16MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00024000
I (65) boot:  1 otadata          OTA data         01 00 0002d000 00002000
I (72) boot:  2 ota_0            OTA app          00 10 00030000 00300000
I (80) boot:  3 ota_1            OTA app          00 11 00330000 00300000
I (87) boot:  4 model            Unknown data     01 82 00630000 00600000
I (95) boot:  5 user             Unknown data     01 82 00c30000 003d0000
I (102) boot: End of partition table
I (107) boot: No factory image, trying OTA 0
I (112) esp_image: segment 0: paddr=00030020 vaddr=3c150020 size=55518h (349464) map
I (183) esp_image: segment 1: paddr=00085540 vaddr=3fc9f6b0 size=06c54h ( 27732) load
I (189) esp_image: segment 2: paddr=0008c19c vaddr=40378000 size=03e7ch ( 15996) load
I (193) esp_image: segment 3: paddr=00090020 vaddr=42000020 size=14c28ch (1360524) map
I (441) esp_image: segment 4: paddr=001dc2b4 vaddr=4037be7c size=13830h ( 79920) load
I (468) boot: Loaded app from partition at offset 0x30000
I (496) boot: Set actual ota_seq=1 in otadata[0]
I (496) boot: Disabling RNG early entropy source...
I (507) opi psram: vendor id : 0x0d (AP)
I (507) opi psram: dev id    : 0x02 (generation 3)
I (507) opi psram: density   : 0x03 (64 Mbit)
I (510) opi psram: good-die  : 0x01 (Pass)
I (515) opi psram: Latency   : 0x01 (Fixed)
I (520) opi psram: VCC       : 0x01 (3V)
I (525) opi psram: SRF       : 0x01 (Fast Refresh)
I (530) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (536) opi psram: BurstLen  : 0x01 (32 Byte)
I (541) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (547) opi psram: DriveStrength: 0x00 (1/1)
I (552) spiram: Found 64MBit SPI RAM device
I (556) spiram: SPI RAM mode: sram 80m
I (561) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (568) cpu_start: Pro cpu up.
I (572) cpu_start: Starting app cpu, entry point is 0x403796d8
I (0) cpu_start: App cpu up.
I (863) spiram: SPI SRAM memory test OK
I (864) spiram: Instructions copied and mapped to SPIRAM
I (992) spiram: Read only data copied and mapped to SPIRAM
I (1033) cpu_start: Pro cpu start user code
I (1033) cpu_start: cpu freq: 240000000
I (1034) cpu_start: Application information:
I (1036) cpu_start: Project name:     willow
I (1041) cpu_start: App version:      0.1.1
I (1046) cpu_start: Compile time:     Oct  3 2023 16:40:47
I (1052) cpu_start: ELF file SHA256:  965498b94f5d7942...
I (1058) cpu_start: ESP-IDF:          v4.4.5-dirty
I (1064) cpu_start: Min chip rev:     v0.0
I (1069) cpu_start: Max chip rev:     v0.99 
I (1074) cpu_start: Chip rev:         v0.1
I (1078) heap_init: Initializing. RAM available for dynamic allocation:
I (1086) heap_init: At 3FCAC990 len 0003CD80 (243 KiB): D/IRAM
I (1092) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (1099) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (1105) spiram: Adding pool of 6464K of external SPI memory to heap allocator
I (1114) spi_flash: detected chip: gd
I (1118) spi_flash: flash io: dio
I (1122) sleep: Configure to isolate all GPIO pins in sleep state
I (1129) sleep: Enable automatic switching of GPIO sleep configuration
I (1136) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1151) spiram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (00:03:06.202) WILLOW/MAIN: Starting up! Please wait...
I (00:03:06.287) WILLOW/MAIN: SPIFFS mounted
I (00:03:06.288) WILLOW/CONFIG: opening /spiffs/user/config/willow.json
E (00:03:06.349) WILLOW/CONFIG: failed to open /spiffs/user/config/willow.json
I (00:03:08.835) WILLOW/NETWORK: initializing SNTP client
I (18:03:08.835) WILLOW/NETWORK: Using configured SNTP server 'pool.ntp.org'
W (18:03:10.599) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (18:03:12.099) WILLOW/WAS: initializing WebSocket client (ws://192.168.1.230:8502/ws)
I (18:03:12.100) WILLOW/WAS: initializing WebSocket client (ws://192.168.1.230:8502/ws)
I (21:01:27.813) WILLOW/NETWORK: SNTP client synchronized time to 1697162487
E (21:01:29.601) WEBSOCKET_CLIENT: Could not lock ws-client within 2000 timeout
E (21:01:29.601) WILLOW/WAS: failed to send WAS get_config message
E (21:01:37.603) esp-tls: [sock=54] select() timeout
E (21:01:37.604) TRANSPORT_BASE: Failed to open a new connection: 32774
E (21:01:37.605) TRANSPORT_WS: Error connecting to host 192.168.1.230:8502
E (21:01:37.613) WEBSOCKET_CLIENT: Error transport connect
I (21:01:37.619) WILLOW/WAS: WebSocket disconnected
E (21:01:57.625) esp-tls: [sock=54] select() timeout
E (21:01:57.625) TRANSPORT_BASE: Failed to open a new connection: 32774
E (21:01:57.627) TRANSPORT_WS: Error connecting to host 192.168.1.230:8502
E (21:01:57.635) WEBSOCKET_CLIENT: Error transport connect
I (21:01:57.641) WILLOW/WAS: WebSocket disconnected
E (21:02:17.647) esp-tls: [sock=54] select() timeout
E (21:02:17.647) TRANSPORT_BASE: Failed to open a new connection: 32774
E (21:02:17.649) TRANSPORT_WS: Error connecting to host 192.168.1.230:8502
E (21:02:17.657) WEBSOCKET_CLIENT: Error transport connect
I (21:02:17.663) WILLOW/WAS: WebSocket disconnected
E (21:02:37.669) esp-tls: [sock=54] select() timeout
E (21:02:37.670) TRANSPORT_BASE: Failed to open a new connection: 32774
E (21:02:37.671) TRANSPORT_WS: Error connecting to host 192.168.1.230:8502
E (21:02:37.679) WEBSOCKET_CLIENT: Error transport connect
```

I'm running WAS on Windows 10 and the firewall is completely disabled. 

Any help would be appreciated!

nikito, 2023-10-13:
Are you able to look at the WAS logs and see what it is showing from its side when the device tries to establish a connection?

kevdliu, 2023-10-13:
I just discovered the issue is caused by a bug on my router firmware that breaks communication between clients on 2.4ghz and 5ghz bands. The computers I tested with were on 5ghz while willow was on 2.4ghz. Sorry for the false bug report and I appreciate your reply nevertheless!

closed 2023-10-13 by kevdliu