HeyWillow/willow #311: Trying to find timeout when calling REST command endpoint.

richardklafter, 2023-10-13:
I am using the REST endpoint to run an LLM and respond with TTS. Everything works great except when the LLM is slow to respond. It seems there is a 5 second timeout somewhere in the stack? I checked the app server's python and didn't see the issue there so perhaps it is in the ESP32? Any help would be appreciated. Happy to pipe it through as a setting in the UI or whatever if I can find the source.

Simple python REST endpoint which demonstrates the issue. 
```
from flask import Flask, jsonify
import time

app = Flask(__name__)

@app.route('/test', methods=['POST'])
def test():
    #time.sleep(4.5) #works
    time.sleep(5) #does not work
    return jsonify("good as gold")

if __name__ == '__main__':
    app.run(host='0.0.0.0', port=3000)
```

nikito, 2023-10-14:
I'm admittedly not super well versed in the C code, but I looked through the endpoint code in Willow and I couldn't find any explicit timeout defined in the code for awaiting the post reply, so I wonder if this may be something inside the espressif libraries..? I'll let one of the other devs chime in here 🙂

kristiankielhofner, 2023-10-14:
@richardklafter is using WAS Command Endpoint support. We're using python requests with the default timeouts. I'm working through a way to handle this and will have it available shortly.

richardklafter, 2023-10-14:
Thanks @kristiankielhofner!

stintel, 2023-12-20:
Timeout in WAS REST Command Endpoint should be 30s as per https://github.com/toverainc/willow-application-server/commit/ff86dd97. Does this answer your question? If yes, please close the ticket. Thanks!

stintel, 2025-02-23:
Question answered and no feedback for > 1 year.

closed 2025-02-23 by stintel