HeyWillow/willow #315: CORS error while flashing

kubanm3, 2023-10-19:
While trying to flash ESP32-S3-Box-3 in console there is error whit cors. Device is succesfully connected with serial port.
Flasher: https://flash.heywillow.io/

[image]

[image]

`Access to fetch at 'https://worker.heywillow.io/api/fetch?url=undefined' from origin 'https://flash.heywillow.io' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource. If an opaque response serves your needs, set the request's mode to 'no-cors' to fetch the resource with CORS disabled.`

Browsers:
Edge: 117.0.2045.60
Chrome 118.0.5993.89

kubanm3, 2023-10-19:
In `Release to flash` dropdown there are no options to choose.
[image]

Maybe that's the reason there is undefined passed as url in request?

kristiankielhofner, 2023-10-19:
Yes, that is the actual issue. @stintel - can you take a look?