HeyWillow/willow #316: HASS Error: HTTP 255

bert269, 2023-10-20:
I am geting an error trying to connect ESP-box to HA.
ESP-BOX screen displays: HTTP Error 255.

This is an authentication error, but I do not know how to fix it. All my other devices are connected to HA, without a problem.
I do not have SSL enabled on my local HA, but I can SSL into in from the outside via duckdsn.org.

The WAS config is therefore setup as HTTP://192.168.1.233:8123 - and for now I am trying to use tovera.io as my WIS (until I get this connection issue sorted out):
[image]

Non-secure connection to HA: [image]

I have regenned the HA token and are certain that it is the correct one - copied / pasted  and viewing it shows it to be correct.

Should I go via duckdsn.org instead? and chage it to HTTPS in WAS? I would hate to go outside my network as I eventually will use WIS and everything should be local.

Ideas on where to start looking to solve this problem?

Thanks

nikito, 2023-10-20:
You need to remove the http from your home assistant host. So it should just be 192.168.1.233 🙂

bert269, 2023-10-20:
darnit - this has BITten me in the past already...Let me try. Thanks.

stintel, 2023-10-23:
We'll add some sanity checks in the WAS UI to prevent such obvious mistakes.

closed 2023-10-23 by stintel