HeyWillow/willow #317: Respeaker support?

jerome83136, 2023-10-21:
Hello,
Great project! Really promising! I will definitely play with this nice toy you made for all of us 😁

I'm already using other software and hardware for voice application and I wonder if there is a chance other hardware will be supported in the future?

So concretely; is there a chance respeaker hardware will be supported ? https://wiki.seeedstudio.com/ReSpeaker/

This hardware is meant for voice applications with embedded voice treatment hardware. Of course this is not esp32 based; so it would require you to release and maintain an arm version of your software.

But there is a big community of arm users in the voice application and home automation workd. So, supporting this kind of hardware would allow you making your public even larger 😉

Thank you for your feedback and if you cannot fulfill my request, please guys fulfill at least this one: never get buyed or killed by GAFAMs 🙏😊

Best regards

kristiankielhofner, 2023-10-22:
Thanks!

We have a generic Willow Linux client that hasn't been released. This will be long but I expect this will be a recurring topic so I want to put the time in to explain our position. My response here will serve as a draft for what we will likely provide on heywillow.io and elsewhere.

Plugging a microphone into a device and making it a voice interface seems obvious and practical in concept/theory. Unfortunately, what I've experienced over the past several years in trying to implement this via miscellaneous hardware, Linux, and software demonstrates it isn't practical in the real-world:

- Linux with all of the realtime configurations and patches available still isn't a realtime operating system like what we use on the ESP-BOX (FreeRTOS). Without the timing guarantees of a realtime operating system you have relatively wide variability just in terms of audio capture and execution with Linux fundamentally, which only gets worse and worse depending on what else the host system is doing. If you look at various projects attempting realtime audio on Linux you will start to understand how difficult and complex this issue is. Sometimes it can be made "good enough" but it often isn't.

- Audio on Linux is really messy in general. ALSA, Pulseaudio, Pipewire... There is no standard configuration across the Linux audio ecosystem and even when you "standardize" on Pulseaudio for compatibility with native Pulseaudio implementations and Pulseaudio via Pipewire there are still issues all over the place: varying buffers throughout the stack, hardware configurations, quirks in underlying drivers, etc. Plain ALSA is tough for other reasons. Then there's JACK which is a completely different animal.

- Varying microphones. Even just the Respeaker series comes in multiple different hardware configurations - i2s, USB, etc that each also have their own quirks due to variations in hardware design and drivers. The Respeaker series has also been more-or-less abandoned in terms of software support and just getting the i2s drivers to compile with your host OS and kernel can be a challenge (I last did this successfully a couple of months ago via some random fork I found on Github).

- Lack of coherent and consistent audio capture. The USB Respeaker alone comes with multiple firmware versions that change the number of output channels. There are also vast underlying differences with varying degrees of configurability in the chipsets some of these devices use (typically provided by XMOS or similar) that introduce even more quirks and variability from the frontend audio processing to the audio levels and format returned by the hardware. There are also varying sample rates supported by the underlying hardware which introduces another rub - audio captured by this hardware often needs to be resampled somewhere and this alone can have the tendency to introduce clicks, pops, varying CPU load, etc. I myself fairly regularly see wake/capture failures because of this with our limited scope and highly optimized software client.

- Doing everything in software. These approaches often don't even support the advanced frontend audio processing Willow does on ESP-BOX (AEC, BSS, VAD, etc) and attempting to do even a subset of this by chaining together software in the audio path introduces even more challenges in quality, variability, latency, and performance. We measure and benchmark performance down to at least the 10s of milliseconds range, which is a tall and nearly impossible order for this approach. The buffering alone is a disaster and I have yet to get our software client to provide the consistent timing we have with the ESP-BOX: variations in timing between wake word, speech command, and any follow-up consistently trips-up this approach. With Willow and an ESP-BOX you can throw wake word and speech/commands at it with nearly any variability in timing and it will do what you expect - even in "rapid fire" scenarios.

- Physical design and packaging. Audio hardware acoustics are a very real thing and without any consistency in terms of enclosures, physical layout, etc between devices there is even more fundamental variance in the audio returned from them.

- Stability and reliability. We routinely do real-world full-stack testing of performance, accuracy, and reliability. Our standard testbench for any change is 1000 loops from wake to TTS feedback on playback from an external audio source. Our maximum acceptable failure rate is 1% and more often than not it's actually 0. We also do fairly regularly do tests simulating a year's worth of voice sessions (10,000 - which is ~27/day across 365 days) while maintaining < 1% failure rates. I have zero confidence a Linux software implementation running on a Raspberry Pi or similar (especially with SD card) can possibly approach this. Without a lot of changes to the operating system to run completely read-only something as simple and routine as a power failure can take out a Linux-based device and even USB audio devices themselves flake out occasionally (because USB). An ESP-BOX is essentially impossible to break (we've never seen or heard of it) and there are other ESP-based devices that have been silently doing their job in challenging conditions for many, many years. This is what we and our users expect with Willow.

[... cut ...] d big tech out; emphasizing private and completely local approaches. I support any path a user takes to achieve that.

On the GAFAM/MAGMA/FAANG point, the use-case and monetization strategy for Willow has always been commercial/enterprise applications where Amazon/Google/etc have no business. Applications like healthcare where the hardware, form factor, management, security, integrations, regulatory environment, economics, etc make the overall approach by these players a fundamental non-starter.

jerome83136, 2023-10-23:
Hello @kristiankielhofner,
Thank you for the very detailed and clear answer.
I clearly understand your point and I think you are 100% right in focusing on your target.

So let's not disturb you anymore with other clients like Raspberry, Respeaker, etc... which are kind of "out of scope" :-)

I just hope Willow will also let some open door for users (like me) who will want to achieve more complex integrations (APIs, MQTT, etc...)

Thank you again for creating this really cool project; I'm impatient of testing it with my (being delivered) ESP BOX ;-)

Best regards

kristiankielhofner, 2023-10-23:
I'm glad my rant didn't scare you off!

Now that we have solid fundamentals - Willow itself, Willow Application Server, Willow Inference Server, new web ui, etc integrations and applications are our next focus! The Willow Application Server is called that for a reason :).

As one example, we have already [started working on MQTT support](https://github.com/toverainc/willow/issues/314) via WAS. Our goal is for WAS to supplement/augment Home Assistant and other supported platforms with voice interface/assistant functionality in WAS via integrations, modules, and applications - including custom user ones. Willow on the ESP-BOX is actually fairly dumb beyond the interface and advanced audio processing. It will end up essentially being a voice terminal interface to WAS which will take on more and more of the heavy lifting for this kind of functionality. Not only does it not make sense to burden a device that should dedicate every available cycle to high quality audio, Python and WAS is much easier and faster to develop for when compared to C on a microcontroller. It's also much easier to work with in terms of upgrades - new functionality can be implemented via WAS without requiring many/any changes to Willow itself (OTA updates for each device, etc). Pull new WAS docker container, done.

We have a general sense from users and our own frustrations regarding the Home Assistant mix of Python code, increasingly complex YAML, etc which we feel is really starting to show some limitations when extended to voice interfaces, assistants, etc. Our intention is to make programming interfaces for this functionality solely available via Python which we feel should be a good mix of approachability, flexibility, and power. We already have some extremely simple and light examples for ChatGPT/self-host LLMs, Google Calendar, e-mail, etc in the works.

We likely won't ever have the approachability and ease of Home Assistant native voice interfaces but our focus is and will continue to be a platform agnostic voice platform for "power users" who are looking for higher performance, higher quality, and more flexibility with emphasis on a completely private and locally hosted approach.

Let us know when you get your ESP-BOX! The good news is the supply shortages should be remedied soon as Espressif has ramped up manufacturing with the ESP-BOX-3 which should be widely available within the next month or so. Going back to the discussion above - the points I made regarding the ESP-BOX are moot if you can't get them :).

X-Ryl669, 2023-10-28:
There are other point to consider. In Rhasspy, the ASR result is run into an (internal) intent matcher, so even with a very bad speech recognition, most of the time, the final intent is recognized well. However, it doesn't integrate well with Home Assistant (it's redundant with HA's own ASR and intent recognition) and doesn't really fit well with continuous audio streaming on the network for the satellite.

Also, on Rhasspy, you can customize the wake word with OWW. This can't be done on the ESP BOX (or not cheap, IIUC).

For example, in HA, fasterwhisper doesn't work at all with french, so we have to rely on DeepSpeech for a working system and only Rhasspy performs well for this.

kristiankielhofner, 2023-10-28:
We have plans leveraging the Willow Application Server (WAS) and the Willow Inference Server (WIS) to apply state of the art matching of commands. With the GPU-focus of WIS we have plenty of options in terms of how to implement this while maintaining the speed and accuracy Willow provides today.

I responded to your other issue regarding custom wake words. If you follow the OWW/HA Github Issues, Discord, Community Forums, etc you will start to see the OWW custom training process doesn't currently work very well in practice, with many users complaining they have to repeat the wake word several times, speak loudly and clearly, etc. There are also many reports of these issues with the wake words provided "out of box". This is is a very hard and fundamental problem to address that applies to all voice interfaces, including commercial solutions and Willow itself. In our view if you have to repeat yourself multiple times, use specific cadence, etc your voice assistant is broken and nearly useless. We've done significant testing with the HA implementation and have experienced this ourselves.

Faster-whisper with the typical models used by HA doesn't work well with French because it doesn't work very well in any language - it's just even worse in French. Running on CPU with tiny is already very slow and it's a very poor quality model. Base is significantly slower and still not very good. With Willow and substantial user feedback over the past six months we have learned that small with beam size 2 should be considered the bare minimum in terms of speech recognition models, even with the drastically superior audio quality provided by Willow and the ESP-BOX.

This is a fundamental problem and you can see from our [benchmarks](https://heywillow.io/components/willow-inference-server/#benchmarks) that even the fastest, most power hungry, and expensive CPUs take 641ms to transcribe 3.8 seconds of speech on a > $1000 CPU (AMD ThreadRipper). We have many users that go "all out" and use Whisper large-v2 with beam size 5, which provides incredible accuracy across a wide variety of languages (including French) for even very complex grammar. Looking at the comparison benchmarks you can see this is still so fast on even the cheapest GPUs like the GTX 1070 in the end it barely matters.

I should note that both faster-whisper and WIS use the same underlying engine and models. The WIS comparison benchmarks on CPU more-or-less match what you get with faster-whisper (WIS is actually slightly faster). The Raspberry PI comparison benchmarks were done with faster-whisper using the exact same parameters HA uses.

The word error rates (WER) provided by [OpenAI tell this story](https://github.com/openai/whisper#available-models-and-languages). Note this chart is for the large-v2 model, with smaller models getting significantly worse as you move down.

Between poor wake word accuracy, transcription, and speed it's faster, easier, and far less frustrating to take your phone out of your pocket and do whatever you need to do there.

Note that while this may seem very critical HA/Esphome and Willow are different projects with different goals and wildly different architectures as a result to meet those goals. There are pros and cons to both projects and we're happy for users to use what works best for them.

closed 2024-03-08 by jerome83136