HeyWillow/willow #330: Feature: Ability to configure custom integrations/APIs other than home assistant

sammcj, 2023-11-12:
It would be _really_ neat if Willow could be configured with custom / user defined integrations.

For example, I want to send the text that willow infers from voice directly to a locally run LLM (AI) endpoint for processing which will return a text response.

At present I can do this by going Willow/WAS/WIS -> Home Assistant -> Home Assistant Local AI / Open AI custom endpoint integration -> LLM service (and back again). Which works pretty well, but adds complexity and points of failure.

Below is an example where I've connected an ESP Box to Willow/WIS/WAS, HA and a large language model (Mistral 7b) running on my home server. It gives me the ability to converse with an AI:

https://github.com/toverainc/willow/assets/862951/0dea21e3-f054-4741-8835-f222cd1e3674

nikito, 2023-11-12:
This is already possible to an extent: In WAS general settings you can use the REST command endpoint 🙂 In the future we plan to introduce logic that will allow for things like Willow -> WIS -> HA/LLM/REST etc -> TTS for example.

stintel, 2025-05-30:
As @nikito explained, there is already basic support for custom APIs. I would recommend to enable "WAS Command Endpoint" in WAS Advanced settings. This way, WAS will be talking to the custom API, and as this is Python, and not C that has to be compiled and flashed on Willow, it's much easier to modify.

closed 2025-05-30 by stintel