HeyWillow/willow #331: Font size is too large for display, text cut off on ESP S3 Box 3

sammcj, 2023-11-13:
https://github.com/toverainc/willow/assets/862951/1a8e61bc-3894-437c-859c-5e7e7045cc91

I went to change the font in the configuration but found that it's hard coded to use a binary file that doesn't seem to have any source / build config:

```
   lv_font_willow = lv_font_load("A/spiffs/user/font/tonnelier.bin");
```

stintel, 2023-12-15:
We've made a change in Willow Application Server to sanitize the response from command endpoints: toverainc/willow-application-server@5f3fe38aa7afc0ecea31dd606bcd60e0f2162be2. If you run WAS with this change included, and enable WAS Command Endpoint in WAS > Configuration > Advanced, this should solve the issue in the video.

stintel, 2023-12-21:
We have just released [WAS 0.2.0-rc.2](https://github.com/toverainc/willow-application-server/pkgs/container/willow-application-server/160552065?tag=0.2.0-rc.2) which includes the aforementioned change. Please test and close this issue if that solves the problem for you.

Thanks!

sammcj, 2023-12-22:
Howdy @stintel, Thanks for looking into this.

Just tested out WAS 0.2.0-rc.2 with both current stable and 0.3.0-rc2 firmware and unfortunately it's no different - the font size is still quite large and scrolls off the screen (very laggy scrolling too?).

sammcj, 2023-12-22:
Actually - to correct myself after another reboot it did update!

It is better at least for short response but if you're getting it to tell you a few sentences the text is still so large it has to scroll for a very long time.

[image]

sammcj, 2023-12-22:
Is the font, typeface size, background colour and basic text layout configurable by the user somewhere?

In a perfect would I'd like to get rid of the 'welcome to willow' and 'i heard:' and simply have what it heard at the top in small text, then the response (without the word response:) across 2-4 lines

stintel, 2023-12-23:
We're an audio device first. We keep the UI very simple because we don't want to spend a lot of time on it. For the same reason, we don't support customizing it, and it's not planned either.