HeyWillow/willow #342: When the "Wake Confirmation Tone" is used, it gets detected as "Ding!"

rarroyo6, 2023-12-13:
When the "Wake Confirmation Tone" gets enabled, it sometimes gets detected as the word "Ding!" immediately after saying the wake word, and it doesn't allow you to continue saying the command.
This doesn't happen all the time (about 30%-50% of the time), lowering the speaker volume helps a bit, but it continues to happen.
Maybe the microphones should be turned off until after the confirmation tone finishes playing.

stintel, 2023-12-13:
#332

rarroyo6, 2023-12-13:
Thank you, I'll try those suggestions when I get home.

rarroyo6, 2023-12-14:
I increased the VAD to 750, and it seems to be working OK now.
I'll try it for a few days and report back if there's still a problem.

I still think that the microphone should be disabled while the speaker is active, since this issue will continue to come up as more people start using Whisper.

Thanks for the help.

stintel, 2023-12-15:
> I still think that the microphone should be disabled while the speaker is active, since this issue will continue to come up as more people start using Whisper.

My gut feeling says this will be too timing sensitive and actually make the experience worse, but we're going to test it locally and make a decision based on those test results.

stintel, 2023-12-15:
You can download a Willow image that mutes the mic before playing wake confirmation tone and unmutes afterwards: https://github.com/toverainc/willow/actions/runs/7223902325

You can OTA update it via WAS following the instructions here: https://github.com/toverainc/willow-application-server#upgrade-with-your-own-willow-builds

As I suspected, mute is too slow and in my testing it's not a good experience at all. However I've never actually used the wake confirmation tone, so I've no previous experience to compare it against. Interested in getting feedback from others.

stiltjack, 2024-02-04:
Same thing here, about 50% of the time. The "Ding" comes so quickly that it is sometimes difficult to get the command in first. Don't know if it is just the way I am speaking, when I am close to the device it seems to hapopen more often.

commit 169e7cbc (2025-02-18): build: switch to upstream ESP-ADF v2.7

This breaks initialization of the touch screen, but we will fix that in
a separate commit, after bumping ESP-IDF to v5.2. We can't bump to
ESP-IDF v5.2 first, because the current ESP-ADF version does not contain
the FreeRTOS patch for ESP-IDF v5.2.

We also have to disable CONFIG_LCD_RGB_ISR_IRAM_SAFE, as this enables
CONFIG_GDMA_ISR_IRAM_SAFE, which results in the following errors during
initialization of the I2S driver:

E (22:15:19.708) gdma: gdma_register_tx_event_callbacks(399): user context not in internal RAM
E (22:15:19.716) gdma: gdma_register_r [... cut]
commit efdd4bc3 (2025-02-19): build: switch to upstream ESP-ADF v2.7

We have to disable CONFIG_LCD_RGB_ISR_IRAM_SAFE, as this enables
CONFIG_GDMA_ISR_IRAM_SAFE, which results in the following errors during
initialization of the I2S driver:

E (22:15:19.708) gdma: gdma_register_tx_event_callbacks(399): user context not in internal RAM
E (22:15:19.716) gdma: gdma_register_rx_event_callbacks(435): user context not in internal RAM

We also need to add a fallback value for ES7210_MIC_SELECT, which is not
defined in the board_def.h file for the ESP32-S3-Box-Lite. Commit
espressif/esp-adf@ebb804bab9beb3e9ef6d1f3a1923aca2b94ae0b7 [... cut]
commit d717a78f (2025-02-19): build: switch to upstream ESP-ADF v2.7

We have to disable CONFIG_LCD_RGB_ISR_IRAM_SAFE, as this enables
CONFIG_GDMA_ISR_IRAM_SAFE, which results in the following errors during
initialization of the I2S driver:

E (22:15:19.708) gdma: gdma_register_tx_event_callbacks(399): user context not in internal RAM
E (22:15:19.716) gdma: gdma_register_rx_event_callbacks(435): user context not in internal RAM

We also need to add a fallback value for ES7210_MIC_SELECT, which is not
defined in the board_def.h file for the ESP32-S3-Box-Lite. Commit
espressif/esp-adf@ebb804bab9beb3e9ef6d1f3a1923aca2b94ae0b7 [... cut]
commit 02ae824e (2025-02-22): build: bump to upstream ESP-ADF v2.7

We need to add a fallback value for ES7210_MIC_SELECT, which is not
defined in the board_def.h file for the ESP32-S3-Box-Lite. Commit
espressif/esp-adf@ebb804bab9beb3e9ef6d1f3a1923aca2b94ae0b7 changed this
to (ES7210_INPUT_MIC1 | ES7210_INPUT_MIC2 | ES7210_INPUT_MIC3) for the
Box and the Box-3, and added a fallback value in es7210.c. Looking at
espressif/esp-adf#1241, it appears MIC3 is used on Box and Box-3 for
AEC. Looking at the schematics, the Box-Lite does not have MIC3
connected, so espressif/esp-adf@ebb804bab9beb3e9ef6d1f3a1923aca2b94ae0b7
corrected [... cut]
closed 2025-02-22 by stintel