HeyWillow/willow #343: TTS response stopped working.

rarroyo6, 2023-12-13:
Started using Willow a few weeks ago with Home Assistant and the default toverainc WIS server, so I could test it before I set-up my own WIS server.
The TTS was working fine at the beginning, it would say the response displayed on the screen, but now it only displays the response on the screen, and it doesn't say it. The "Wake Confirmation Tone" is played, so the speaker, volume, etc. is working on the client.
I'm not sure if it happened after one of the updates on the client, or if the toverainc server stopped processing TTS for some reason.

nikito, 2023-12-13:
Just tested the tovera public server and am able to generate TTS, so that appears to not be the issue.
Have you checked your settings to make sure the response type didn't accidentally get set to Silence? Also, have you tried physically unplugging your willow devices for ~10 seconds, then plugging back in and trying again? I have noticed that in some rare cases something can happen where the device gets in a state where audio won't play or is garbled, and that seems to correct it.

stintel, 2023-12-13:
If @nikito's suggestion doesn't help, please provide the following data:
* Willow version
* serial console logs

rarroyo6, 2023-12-14:
Thank you, the response type was set to Silence.

closed 2023-12-14 by rarroyo6