HeyWillow/willow #354: i2s is invalid on ESP32 Lite

yohaskan, 2023-12-30:
flash seem ok and Home assitant respond il log file, but no audio, no chime sound come from ESP32 lite

In log of Willow Console 
```
E (00:44:41.920) i2s(legacy): i2s_check_cfg_validity(926): this i2s port is in use
E (00:44:41.928) i2s(legacy): i2s_driver_install(1582): I2S configuration is invalid
```

nikito, 2023-12-31:
Try to completely unplug the box lite for several seconds, then plug it back in? Also was this via Web flasher?

nikito, 2023-12-31:
Also, have you ensured you have enabled tts in your General Settings in WAS and applied everywhere?

stintel, 2024-01-02:
> E (00:44:41.920) i2s(legacy): i2s_check_cfg_validity(926): this i2s port is in use
> E (00:44:41.928) i2s(legacy): i2s_driver_install(1582): I2S configuration is invalid

This error message comes from ESP-ADF, because it's still using the legacy I2S driver, which does not like being installed twice. This is completely harmless, yet unfortunate as it keeps causing people to think this is a problem. I test regularly on all supported devices, and while I currently cannot test a Box-Lite, I can confirm this error message appears on the original Box, and the Box-3, yet everything works fine.

As @nikito suggested, make sure you have actually set _Willow Audio Response_ in WAS to _Chimes_ or _Text-to-Speech_. Alternatively, try the _Locate Device_ feature in the _Clients_ tab.

commit ce52787a (2024-01-02): audio: only install I2S driver once

This requires an ESP-ADF bump.

Closes: #125
Closes: #354
commit 915dc272 (2024-01-02): audio: only install I2S driver once

This requires an ESP-ADF bump.

Closes: #125
Closes: #354
commit 3d374ac1 (2024-01-02): audio: only install I2S driver once

This requires an ESP-ADF bump.

Closes: #125
Closes: #354
PR #357 (2024-01-02): audio: only install I2S driver once
This requires an ESP-ADF bump.

Closes: #125
Closes: #354
closed 2024-01-02 by kristiankielhofner