HeyWillow/willow #364: {   "response": "Chatbot not installed or supported" }

TRONBFFL, 2024-02-14:
I'm not sure what I did wrong in the setup. 
I dont see anywhere where it tried to download and cache model though. 
I changed the settings.py support chatbot true - am I missing something?

wis-1    | [2024-02-14 16:11:54 +0000] [94] [DEBUG] FASTAPI: Got chatbot request with text: hello there
wis-1    | [2024-02-14 16:11:54 +0000] [94] [WARNING] CHATBOT: Not installed or supported
wis-1    | [2024-02-14 16:11:54 +0000] [94] [DEBUG] FASTAPI: Got chatbot response with text: Chatbot not installed or supported

nikito, 2024-02-15:
If you are running latest WIS server, chatbot was removed, so these are likely pointing to that fact. 🙂

nikito, 2024-02-15:
Adding here, i realized i was thinking of the split_arch branch, it appears main still has support for chatbot (but this will go away soon in favor of using some of the more refined engines out there). You are potentially seeing this error because the card architecture/capability is too low or vram is too low.