HeyWillow/willow #371: Feature Request: Support Esp32-s3 with Microphone and Speaker

bwims, 2024-04-23:
Given the expense and lack of availability of the Esp32-S3-Box coupled with the particular purpose of HA voice assistant, would you please consider supporting a combination of Esp32-s3 with an attached microphone and speaker?  Such support being to recommend of boards and peripherals that ought to work.

Thanks!

B.

nikito, 2024-04-23:
I don't think there is any particular microphone/speaker we could recommend, as it just depends on what is supported by the ESP-ADF. In our discord we have some users working on some custom hardware/PCBs for this, but I believe it is still a work in progress. It also requires writing new firmware for Willow to support this device, which the community is also working on from what I understand.
Espressif is supposed to be ramping up production more, and units are evidently expected to start coming in again in July from what I see on a few sites. Espressif also just acquired M5Stack, so we may potentially see more hardware from that as well. :)

bwims, 2025-02-17:
Closed because I have one of these now.

https://thepihut.com/products/home-assistant-voice-preview-edition

[image]

closed 2025-02-17 by bwims

nikito, 2025-02-17:
Additionally, the main dev has unfortunately passed away, so updates to this repo will likely stop. :/