HeyWillow/willow #372: command not found: ip

taozhiyuai, 2024-05-11:
`taozhiyu@603e5f4a42f1 ~ % ip route get 1.1.1.1 | grep -oP 'src \K\S+'

zsh: command not found: ip
grep: invalid option -- P
usage: grep [-abcdDEFGHhIiJLlMmnOopqRSsUVvwXxZz] [-A num] [-B num] [-C[num]]
	[-e pattern] [-f file] [--binary-files=value] [--color=when]
	[--context[=num]] [--directories=action] [--label] [--line-buffered]
	[--null] [pattern] [file ...]`

my laptop is MAC 14.4. the error info is show above. anyone can help?

taozhiyuai, 2024-05-11:
I execute this command instead.

taozhiyu@603e5f4a42f1 ~ % ifconfig | grep 'inet ' | grep -v 'inet6 ' | awk '{print $2}'
127.0.0.1
192.168.31.110

taozhiyuai, 2024-05-11:
I execute this command instead.

taozhiyu@603e5f4a42f1 ~ % ifconfig | grep 'inet ' | grep -v 'inet6 ' | awk '{print $2}'
127.0.0.1
192.168.31.110

http://127.0.0.1:8502, http://192.168.31.110:8502 

both can not be accessed.

nikito, 2024-05-12:
This is potentially an issue with docker not passing traffic out to the host. How are you running the container?

taozhiyuai, 2024-05-12:
> This is potentially an issue with docker not passing traffic out to the host. How are you running the container?

1.install docker on my MAC laptop.

2.docker run --detach --name=willow-application-server --pull=always --network=host --restart=unless-stopped --volume=was-storage:/app/storage ghcr.io/toverainc/willow-application-server

3.ip route get 1.1.1.1 | grep -oP 'src \K\S+'

nikito, 2024-05-12:
I think others have found that docker on Mac does not work with network=host, i think you have to use network=bridge

taozhiyuai, 2024-05-12:
> I think others have found that docker on Mac does not work with network=host, i think you have to use network=bridge

>>>>>TERMINAL LOG

taozhiyu@603e5f4a42f1 ~ % docker run --detach --name=willow-application-server --pull=always --network=bridge --restart=unless-stopped --volume=was-storage:/app/storage ghcr.io/toverainc/willow-application-server
latest: Pulling from toverainc/willow-application-server
Digest: sha256:06c0310585f2defa13e0f02922fa0075fbaf14bf5e63f1437596716c041e019e
Status: Image is up to date for ghcr.io/toverainc/willow-application-server:latest
cda8c2a48ec77e011d98d7aae0d70db913f05cefca8e744aada82806baffff62
taozhiyu@603e5f4a42f1 ~ % ifconfig | grep 'inet ' | grep -v 'inet6 ' | awk '{print $2}'
127.0.0.1
192.168.31.110

>>>>>> DOCKER LOG
2024-05-13 06:09:45 [2024-05-12T22:09:45] INFO:     Started server process [9]
2024-05-13 06:09:45 [2024-05-12T22:09:45] INFO:     Waiting for application startup.
2024-05-13 06:09:45 [2024-05-12T22:09:45] INFO:     Application startup complete.
2024-05-13 06:09:45 [2024-05-12T22:09:45] INFO:     Uvicorn running on http://0.0.0.0:8502 (Press CTRL+C to quit)

taozhiyuai, 2024-05-12:
### network=host

>>>> TERMINAL LOG

taozhiyu@603e5f4a42f1 ~ % docker run --detach --name=willow-application-server --pull=always --network=host --restart=unless-stopped --volume=was-storage:/app/storage ghcr.io/toverainc/willow-application-server
latest: Pulling from toverainc/willow-application-server
Digest: sha256:06c0310585f2defa13e0f02922fa0075fbaf14bf5e63f1437596716c041e019e
Status: Image is up to date for ghcr.io/toverainc/willow-application-server:latest
4e7eef326ef825af52ebdf6d577c2b7dc77cef80c21ae1ad6638649246705aa8
taozhiyu@603e5f4a42f1 ~ %  ifconfig | grep 'inet ' | grep -v 'inet6 ' | awk '{print $2}'
127.0.0.1
192.168.31.110

>>>>DOCKER LOG

2024-05-13 06:15:49 [2024-05-12T22:15:49] INFO:     Started server process [9]
2024-05-13 06:15:49 [2024-05-12T22:15:49] INFO:     Waiting for application startup.
2024-05-13 06:15:50 [2024-05-12T22:15:50] INFO:     Application startup complete.
2024-05-13 06:15:50 [2024-05-12T22:15:50] INFO:     Uvicorn running on http://0.0.0.0:8502 (Press CTRL+C to quit)

taozhiyuai, 2024-05-12:
both not work on my MAC

nikito, 2024-05-12:
I'm not sure what specific issue is occurring on mac, our Linux users use the original command you tried without issue. Maybe try running the docker the same way you run the others? Perhaps excluding the network argument may help? 
Additionally that network mode won't change anything in the logs, it just affects how the container's network is exposed through docker. Regardless of the argument, the container itself always listens on 0.0.0.0, which means all interfaces. 🙂

taozhiyuai, 2024-05-13:
> I'm not sure what specific issue is occurring on mac, our Linux users use the original command you tried without issue. Maybe try running the docker the same way you run the others? Perhaps excluding the network argument may help? Additionally that network mode won't change anything in the logs, it just affects how the container's network is exposed through docker. Regardless of the argument, the container itself always listens on 0.0.0.0, which means all interfaces. 🙂 Another thought, what ip do you use to talk to the other containers? That ip would ultimately be the same ip this would listen on, since that is the docker Host's ip.

oneapi, dify or other docker container, I use localhost:port