HeyWillow/willow #374: Trailing slash in openHAB URL leads to error

happenpappen, 2024-05-23:
When selecting "openHAB" as command endpoint in WAS and setting the openHAB URL to "https://<fqdn>:8443/" the client will call the rest api with URL "https://<fqdn>:8443//rest/..." like in this example:

`I (05:27:42.682) WILLOW/OPENHAB: sending 'Light on the bathroom.' to openHAB REST API on 'https://example.org:8443//rest/voice/interpreters'
I (05:27:43.198) WILLOW/HTTP: HTTP status='405' content_length='618'
`

which will lead to an error "POST is not supported by this URL...". When i remove the trailing slash, the call will succeed.

Don't know, if this is more an openHAB or a Willow issue, but i think the double slash should be handled anywhere...

nikito, 2024-05-23:
The issue is that technically, double slashes are allowed by the URL spec and normally it should work without issue, but maybe openHab doesn't like it. We could probably add some validation in the UI to not allow the user to enter a trailing slash. :)

stintel, 2025-02-23:
This is documented in the README. See #110. If you want to see it fixed, please report the issue to openHAB.

closed 2025-02-23 by stintel