HeyWillow/willow #376: Device Volume too low

ccsmart, 2024-05-30:
First up, my kudos for this project.
A issue i have is with the ESP32-S3-BOX-3 devices speaker volume, which is way too low, as in noticeable only if the room is dead silent. Web research seems to indicate that its not specific to willow. The question how to increase speaker volume comes up eg on esphome as well. I tested with two brand new devices which i flashed with willow first thing (ie before going through preinstalled demo, whatever). The two have an ever so slight difference in volume.
Im thinking the setting in WAS config server might be a relative "channel" volume to some other "total/overall" volume the latter of which would then be very low.
Whatever the cause, despite research, i am unable to get the volume up such that for all practical means, there is no audio.
Maybe somebody has a hint... ?
Thanks.

nikito, 2024-05-30:
Tbh even if that were the case, i don't think the shaker will ever reach a volume level that users find "practically usable", ie on par with Google home or Alexa. Those devices have much larger and more purpose built speakers whereas the speaker in the esp box is tiny. On the esp box i have to keep my volume at 90 or else the audio actually distorts, and even at 90 while it is loud it's not as loud as a normal speaker.

ccsmart, 2024-05-30:
> Tbh even if that were the case, i don't think the shaker will ever reach a volume level that users find "practically usable", ie on par with Google home or Alexa. ...

unrelated isn't it ?

nikito, 2024-05-30:
Not sure what you mean? It has audio, and it's audible, but no amount of software changing is gonna fix the physical limitations of the speaker, that'd something espressif would have to address with the box 3. That said users also do mods where they create their own s3 devices and add their own audio dac/speakers and such, that just isn't supported or if the box with the s3.

ccsmart, 2024-05-30:
Nope: "way too low, as in noticeable only if the room is dead silent" is not about the physical limitations.
What the actual cause is is unclear. But eg the "location" alert is quite audible. No tin the sense of a firealarm, but perfectly ok for the size of the device. Ordinary people can make an educated guess on that topic.

And the "for comparison: https://github.com/esphome/firmware/issues/139" section also does not speak about fixing hardware. In their case it was an issue with a newer version of the idf.. or the use of that. Issue tbd. Either way, fixable by software.

Dont get me wrong i'm not trying to make willow responsible, nor their team: "Maybe somebody has a hint... ?" Thats what i'm asking for here in a project i'm eager to get into (as a user) and provide feedback and that's likely gathering similar minded people with probably similar problems. If this would be the place to find that answer to that shared issue on google... sure a great thing.

So if this is a place where audio volume on ESP32-S3-BOX-3 can / should be discussed, let it happen. If not, just close it and i'll look somewhere else.

nikito, 2024-05-30:
Yeah no problem, just trying to understand what is being described, as mine is so loud it distorts sometimes. I'd put a clip but gh only allows 10mb and the video is 98 mb evidently. 😄
Are you saying yours is so quiet that even in a silent room you can't hear it? If so that's sounds like a different problem, as we have plenty of users that use the audio and have not reported this. Could be defective hardware even 🙁

nikito, 2024-05-30:
Video example:
https://photos.app.goo.gl/dZn3QMKvL5ZwfLTz9

ccsmart, 2024-05-30:
"Are you saying yours is so quiet that even in a silent room you can't hear it?"

There is a chance to notice that *something is being said*. But there is no chance that you'd be able to hear enough to understand what was said.

2x defective HW while the "locate my device" alert is audible.. possible but unlikely. Also, others see similar problems in principle. So i'm thinking it may be along the lines of https://github.com/esphome/issues/issues/5791#issuecomment-2123245127

nikito, 2024-05-31:
Ah i actually saw that issue playing with home assistant. Thing is though as far as i am aware willow isn't using that version of esp idf.

nikito, 2024-05-31:
One other thought, are you flashing via Web flasher or by compiling and flashing locally?

ccsmart, 2024-05-31:
Flashed using web flasher.
I'm not sure about it but iirc esphomes case is such that if you stay <= 4.4.6 they are ok and they haven't arrived at 5.X yet. Willow has reached >4.4.6 in September 2023. I'm running a local WIS instance. Would the system work with a Willow compiled of September 23 IDF <=4.4.6 to test ? ANd if yes how would i do that ?

ccsmart, 2024-05-31:
Heres some magic... It just started working. But dont ask me what made it... I did not change volume setting. If anything i was playing around with custom_settings.py. Trying to change language - and back. sv on/off, chat on.

ccsmart, 2024-05-31:
Seems magic question is resolved.
A total shutdown / power out of the device was necessary.
On one of the two, the half charged battery died overnight. That one started working when it was recharged.
The other one i just physically disconnected for an unrelated cause. After starting back up it started working.
For clarification... i did multiple resets and such in between. Those did not help.
So it seems it is necessary to: flash, configure, restart, really power off (dead) and restart.

nikito, 2024-05-31:
We did see that before, some weird quirk with the boxes. I usually suggest that to users having trouble, didn't cross my mind this time 😅

stintel, 2025-02-22:
Maybe we can print a message in Willow Web Flash to instruct the user to power-cycle the device after flashing?

stintel, 2025-02-25:
> Maybe we can print a message in Willow Web Flash to instruct the user to power-cycle the device after flashing?

Hah, looks like there's a PR open for that already: toverainc/willow-web-flash#15.

stintel, 2025-04-03:
We had another report of no or very low volume after initial flash, and another confirmation a power-cycle fixed it. I've merged the PR to Willow Web Flash that instructs people to do a power-cycle after flash. Closing.

closed 2025-04-03 by stintel