HeyWillow/willow #382: generated_cmd_multinet.h: No such file or directory

hursto75, 2024-11-04:
./utils.sh build
Getting the error when i try to run the build for the hardware bin image.

Files/__idf_main.dir/audio.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/audio.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/audio.c.obj -c /willow/main/audio.c
/willow/main/audio.c:47:10: fatal error: generated_cmd_multinet.h: No such file or directory
   47 | #include "generated_cmd_multinet.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

thess, 2024-12-12:
This file is created by the python script located in the `speech_commands` folder. The file is only included if **Multinet** support is enabled by `WILLOW_SUPPORT_MULTINET` is defined (default in CMakeLists.txt). If `generate_commands.py` is not successful, no file is created. See `generate_speech_commands()` in `utils.sh`. If you are trying to get **Multinet** working, I'd be interested in how you enabled it in WAS so that this file is automatically generated.

Best to just disable `WILLOW_SUPPORT_MULTINET` in `CMakeLists.txt`. If you want **Multinet** support, you will have to create this file somehow.

stintel, 2025-01-23:
You can work around it by running like this inside the build container:

```
./utils.sh clean
CI=1 ./utils.sh build
```

commit 4f8803e8 (2025-02-18): build: make MultiNet support opt-in

We haven't supported MultiNet in any of the Willow or WAS releases.
Let's make it opt-in, rather than disabling it in CI builds. This will
solve the missing generated_cmd_multinet.h file for people trying to
build manually.

It is still possible to enable MultiNet when building manually by
setting the MULTINET environment variable, but we will not provide
support for any issues related to it. Issues will be closed without
resolution.

Closes: #382
PR #392 (2025-02-18): build: make MultiNet support opt-in
We haven't supported MultiNet in any of the Willow or WAS releases. Let's make it opt-in, rather than disabling it in CI builds. This will solve the missing generated_cmd_multinet.h file for people trying to build manually.

It is still possible to enable MultiNet when building manually by setting the MULTINET environment variable, but we will not provide support for any issues related to it. Issues will be closed without resolution.

Closes: #382
commit 99f44c2f (2025-05-29): build: make MultiNet support opt-in

We haven't supported MultiNet in any of the Willow or WAS releases.
Let's make it opt-in, rather than disabling it in CI builds. This will
solve the missing generated_cmd_multinet.h file for people trying to
build manually.

It is still possible to enable MultiNet when building manually by
setting the MULTINET environment variable, but we will not provide
support for any issues related to it. Issues will be closed without
resolution.

Closes: #382
closed 2025-02-18 by stintel