HeyWillow/willow #385: hosted willow inference server

soulripper13, 2024-12-06:
Is the hosted inference server still valid?
https://infer.tovera.io/api/willow

stintel, 2025-02-18:
See https://github.com/toverainc/willow-inference-server/issues/161. I suspect that due to Kris' passing, the data center where the public WIS server was hosted took it down.

stintel, 2025-04-01:
The public Willow Inference Server was never intended for people to use permanently, but to give them an easy way to test Willow without having to set up their own WIS. As I do not have the funds to bring up a public WIS myself, and there is a usable WIS replacement using a CloudFlare Worker (https://github.com/JayPearlman/WIS-CF-Worker), a new public WIS is unlikely to happen anytime soon. The recommended way is still to run your own WIS, but for the people who are unable to do so, the alternative exists.

closed 2025-04-01 by stintel